acd/infinitive #1: default build issue

nebulous, 2016-10-20:
My Gofoo is weak, but I can't seem to get infinitive to compile (on an arm5 board) due to a missing function reference. See below for an example.
_it does compile with the StaticFS line commented out as a test_

```go get github.com/acd/infinitive
go/src/github.com/acd/infinitive/webserver.go:27: cannot convert nil to type bool
go/src/github.com/acd/infinitive/webserver.go:74: r.StaticFS undefined (type *gin.Engine has no field or method StaticFS)```

acd, 2016-10-20:
That's odd.  I just tried a clean build straight from github and it worked fine for me on my rPi.  Based on the errors you're seeing my guess is that you might be building against an old version of [gin](https://github.com/gin-gonic/gin), though `go get` should always grab the latest from what I understand.

If you don't mind digging in a bit, look at the function definition for `Bind()` in `src/github.com/gin-gonic/gin/context.go` within your `$GOHOME`.  I suspect it's returning a `bool` instead of an `error`.  FWIW, the SHA1 sum of `context.go` that `go get` pulled from github on my system is `0692e659b5f9b5e0de02e3600f95c29e27384d3d`.

nebulous, 2016-10-20:
Ah, yep. I had an older version of gin already pulled. Probably not a problem many people will run into. Unfortunately Infinitive won't build on this limited board so I'll need to cross compile it. Go makes that crazy easy as I recall, so I'll give it a shot later on.

closed 2016-10-20 by nebulous

nebulous, 2016-10-20:
Cross compiled for armv5 just fine 👍