acd/infinitive #2: No heatActive and coolActive data

Will1604, 2016-11-15:
I found the comment in infinitive.go about heatActive and coolActive not being in the mode field. I'd like to log these so am looking into where we can get this information from.

I'm thinking of logging the state of the unknown fields in TStatCurrentParams and TStatZoneParams to see if anything changes there and correlates with what's going on in the system. Does that sound worth the effort - or are these fields known not to contain anything useful?

Perhaps nothing from the tstat can tell whether heat or cool is on, or at what level. Perhaps we need to look in the responses from the heat pump. Does anyone have any clues?

nebulous, 2016-11-15:
I haven't bothered to decode these on my system, but if it helps get you 
started, bit fields seem to be stored at 00 3E (02|03|04) on my heatpump 
which correlate to regular changes in its status:

And writes to the fancoil seem to toggle its status fairly predictably 
as well:

nebulous, 2016-11-15:
I haven't bothered to decode these on my system, but if it helps get you 
started, bit fields seem to be stored at 00 3E (02|03|04) on my heatpump 
which correlate to regular changes in its status:

[image]

And writes to the fancoil seem to toggle its status fairly predictably 
as well:

[image]

acd, 2016-11-16:
I assume there's some way to determine the active mode/stage from the thermostat but I haven't been able to isolate it.  The bits that I thought indicated heat/cool active instead seem to be set whenever the HVAC system is running and aren't tied to heating vs. cooling.  They could indicate the currently active stage or something along those lines.  

Logging changes in the fields is probably the best way to go.  Infinitude has a very nice web UI for passively observing changes in data exchanged on the ABCD bus.  That might be useful, although not for the specific scenario you described as it requires data retrieved by polling the thermostat.  I actually have some uncommitted code I added to infinitive that logs when changes are observed to table data.  If you're interested I can commit that change if you think it would be helpful with your spelunking.

nebulous, 2016-11-16:
It may be worth noting that [brybus](https://github.com/3tones/brybys) also has an sql logger, and the cardump utility that comes with Infinitude outputs tsv such that you can pipe it into a file (or directly into sqlite). We really should work out a standard for stream logging and *especially* for register value definitions. There was some talk on [this cocoontech thread](http://cocoontech.com/forums/topic/11372-carrier-infinity/?p=222943) starting around post #210 around that subject which fizzled out a bit.

Will1604, 2016-11-16:
Spent some time yesterday getting more familiar with the Infinitive code and a broader sense of the exchanges on the bus.

Began working up some code to snoop the fan coil. I'm using the code for the air handler snoop as a pattern for this. In fact there may be very little to do to get something useful. I can see the thermostat is reading the same registers from the fan coil that you're using to obtain blower speed and air flow from the air handler (0x000306 and 0x000316). Does it seem a reasonable assumption that the fan coil would also be providing  blower speed and air flow? Does the response for a given register have same structure irrespective of the device it's coming from?

However, looking at the data coming back from the coil in those two registers. I wasn't seeing any change over the period I monitored. Not yet sure that I've got an adequate sample set of bus data though.

I'm finding the documentation of the protocols on the Infinitude wiki useful
https://github.com/nebulous/infinitude/wiki/Infinity---interpreting-data
"Register" and "table row" are effectively synonymous
The two examples above are different rows of the same table - table 03 - RLCSMAIN

I'd like to get Infinitude up and running. It seemed to rely on inserting itself in the http stream from a Touch thermostat - which I don't have. Perhaps I misunderstood. I'll look again at how to configure it to get the serial data.

Will1604, 2016-11-17:
The code to capture the blower speed and air flow worked for the fan coil. I got reasonable numbers back for blower speed and duct airflow.
Just changing the maxSrc parameter on snoopResponse should be all that's needed. 
infinity.snoopResponse(0x4000, 0x42ff, func(frame *InfinityFrame)

Will1604, 2016-12-16:
Created a fork and committed code which decodes stage from the thermostat. On a heat pump system with 10kW electric heat the stage seems to correspond to
1 - heat pump low
2 - heat pump high
3, 4 - electric heat on. Seems like a reasonable guess that this corresponds to the two levels of electric heat available on some 10kW heaters - but have not confirmed that.

closed 2016-12-16 by Will1604

acd, 2016-12-17:
I'll give this a shot on my system to see what the numbers look like.  I have a heat pump and a gas furnace (what Carrier calls a dual fuel system).  It's so cold here that the system is only running the furnace these days.

Once you've got a reasonable degree of confidence in your interpretation of the data feel free to submit a pull request.

Will1604, 2016-12-17:
One catch is that those stage values are dependent on other settings in the system, If I set the thermostat to "Electric Heat Only" (under Advanced:Heat Source) then stage reads 1 and 2 for electric heat.

The Heat Pump responses back to the thermostat in register 003e02 are useful. 04 is high heat and 02 is low heat.

The fan coil (4201) seems to be indicating electric heat in the high byte of register 000316. 03 seems to correspond to electric heat on. It would get good to know what you see from the furnace in that byte.

I'll add code those to show data from those responses via the API and publish into my fork.