acd/infinitive #5: Bryant Evolution compatibility

TheCase, 2017-01-15:
Tried this on my Evolution system today - I don't think it spits out the same codes.  All the outputs on the web read as zero. Here's a link to a section of the log output: https://gist.github.com/TheCase/ba86e6bd7a9587b3217bba41f4718ebf

[image]

acd, 2017-01-15:
I've heard from others that it does work on some Bryant systems.  Your results match the same behavior reported by a user with a Carrier non-WiFi touchscreen thermostat.  If you have the equivalent Bryant thermostat then that would explain it.  I'm unsure why Infinitive doesn't work with your setup.  I don't have a capture of a SAM interacting with your model of thermostat so I haven't been able to figure out what's different.

TheCase, 2017-01-16:
I have the wifi version of the Bryant. I'm curious if the lack of usable data has something to do with possible communication errors.

acd, 2017-01-16:
I don't see any communication errors in the logs you provided.  Occasional timeouts are expected.  From what I can see you have the RS-485 adapter connected to the bus properly.  Your thermostat is simply not returning any data when queried for the current operating settings.

acd, 2017-01-18:
Closing this issue.  Can't add support for the newer generation of thermostats without a SAM capture of the bus interaction.

closed 2017-01-18 by acd

TheCase, 2017-01-18:
If that is something I can help out with and provide via the RS485 comms, let me know.

bvarner, 2017-02-18:
FWIW - I have a Bryant system with wifi touch thermostat (SYSTXBBECW01).

My log contains many lines like this:
```
INFO[0115] read frame: 2001 -> 9201: RESPONSE 003b03000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

I assumed it was my homebrew pi->485 hardware.

TheCase, 2017-02-18:
@acd is it possible for me to get you a SAM capture, or does that require additional hardware?

acd, 2017-03-01:
A SAM capture requires someone with a touch thermostat and a SAM to capture the ABCD bus communication.

ehwest, 2017-03-04:
I assume the issue is the registers (table/row) data is just a tad different.
This data HAS to be there somewhere.
But I'm just not clear how you decode temperatures and the "F" following.  If I knew this, I'd search for that in the outputs of the scandevtable.py or something.

acd, 2017-03-04:
Infinitive uses the tables defined in `tables.go` to interact with the supported (older) thermostats.  Setpoint and current temperature each represented using a single byte.  The structs map directly to the table component of wire frames, so you can determine offset by looking at the field position in the struct if necessary.

bvarner, 2017-03-04:
Where did this 'table' parlance come from? 

I have a sneaking suspicion that @acd has nailed it in that the actual embedded controllers are dumping out 'structs' to the 485 bus, just like infinitive does.

acd, 2017-03-04:
I've been using the terminology first coined on the [Cocoon Tech thread](http://cocoontech.com/forums/topic/11372-carrier-infinity/).  There's some good information contained there, in particular one post that has some of the table definitions.  I'm not sure how it was obtained and the poster isn't active on the site anymore.

ehwest, 2017-03-04:
OK, I think the problem is the code as is, is trying to get the THERMOSTAT data from the FURNACE, which is responding with all zeros.  The trace clearly shows this.  To get the THERMOSTAT data, you'd try to READ from the SAM (address 0x9201) which isn't on the network, but doesn't matter,  and ask the THERMOSTAT (address 0x2001) for its data.   As it is, nobody is asking the THERMOSTAT for its data, certainly not the furnace.

acd, 2017-03-04:
What are you talking about?  What problem?  Infinitive, impersonating a SAM, is asking the thermostat for two tables that correspond to operating parameters.  The touchscreen thermostats respond with all zero bytes for that table.  The older non touch thermostats work fine.  There is no problem, instead the issue is trying to interact with a new device without having a working capture to use for guidance.

ehwest, 2017-03-04:
The scandevtable.py tool shows SAMINFO type data at tables 003b20  not 003b02.  That might be interesting.

acd, 2017-03-04:
Earle, I'm not sure what you're trying to accomplish with this thread but a github issue is not the proper forum to brainstorm about what you think my code might be doing wrong on the newer hardware.  You're welcome to play around with it, resolve any issues, and submit a pull request.  I'm done responding to the random inaccurate stuff you're throwing out here.