acd/infinitive #6: Link to Coontech broke in README

ehwest, 2017-02-28:
Link to Coontech broke.

Will1604, 2017-02-28:
Works for me.
http://cocoontech.com/forums/topic/11372-carrier-infinity/

ehwest, 2017-02-28:
Very sorry....Buried in the README there is this.  These links go nowhere.
OK, try this.

www.hvac-talk.com/vbb/attachment.php?attachmentid=3061&d=1185394256 

http://www.xpedio.ca...11-20156-25.pdf

acd, 2017-03-01:
Not sure what you're talking about; those links are not included in any of the Infinitive documentation.  I am closing this issue.

closed 2017-03-01 by acd

Will1604, 2017-03-01:
Those links were in a post on the Cocoontech forum. Looks like the poster was posting documents for the SAM. I've found this previously. I think it's the same document.
http://gpsinformation.info/joe/carrier/InfinityThermostatSpecifications.pdf
Describes the protocols a home automation system would use to interface with the SAM using RS-232.

ehwest, 2017-03-01:
Thats it, Will1604 Thanks!