acd/infinitive #7: Blank UI

ehwest, 2017-03-01:
Using PI 2B on a wifi network. The basics seem to be working...see log below

[image]

```
pi@raspberrypi:~/infinitive$ ./infinitive -httpport=8080 -serial=/dev/ttyUSB0 1> weblog.txt 2> log.txt
 
pi@raspberrypi:~/infinitive$ cat log.txt
time="2017-02-28T19:24:02-05:00" level=info msg="opening serial interface: /dev/ttyUSB0" 
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:   export GIN_MODE=release
 - using code:  gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /api/zone/1/config        --> main.webserver.func1 (3 handlers)
[GIN-debug] GET    /api/zone/1/vacation      --> main.webserver.func2 (3 handlers)
[GIN-debug] PUT    /api/zone/1/vacation      --> main.webserver.func3 (3 handlers)
[GIN-debug] PUT    /api/zone/1/config        --> main.webserver.func4 (3 handlers)
time="2017-02-28T19:24:02-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b03" 
[GIN-debug] GET    /api/ws                   --> main.webserver.func5 (3 handlers)
time="2017-02-28T19:24:02-05:00" level=info msg="read frame: 2001 -> 5001: READ     000304" 
time="2017-02-28T19:24:02-05:00" level=debug msg="transmitting frame: 200192010300000b003b03ecba" 
[GIN-debug] GET    /ui/*filepath             --> github.com/gin-gonic/gin.(*RouterGroup).createStaticHandler.func1 (3 handlers)
[GIN-debug] HEAD   /ui/*filepath             --> github.com/gin-gonic/gin.(*RouterGroup).createStaticHandler.func1 (3 handlers)
[GIN-debug] GET    /                         --> main.webserver.func6 (3 handlers)
[GIN-debug] Listening and serving HTTP on :8080
time="2017-02-28T19:24:02-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b03000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:02-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b02" 
time="2017-02-28T19:24:02-05:00" level=debug msg="transmitting frame: 200192010300000b003b022d7a" 
time="2017-02-28T19:24:02-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b020000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:02-05:00" level=info msg="read frame: 2001 -> 5001: READ     000504" 
time="2017-02-28T19:24:02-05:00" level=info msg="read frame: 5001 -> 2001: RESPONSE 000504032003e802d003d4" 
time="2017-02-28T19:24:03-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b03" 
time="2017-02-28T19:24:03-05:00" level=debug msg="transmitting frame: 200192010300000b003b03ecba" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b03000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:03-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b02" 
time="2017-02-28T19:24:03-05:00" level=debug msg="transmitting frame: 200192010300000b003b022d7a" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b020000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 2001 -> 4001: READ     000404" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 0004041078000f1b1f030202570866084509950000" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 2001 -> 4001: READ     00040d" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00040d000000000000" 
time="2017-02-28T19:24:03-05:00" level=info msg="read frame: 2001 -> f1f1: WRITE    000202131802" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> f1f1: WRITE    0002031c0211" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: WRITE    00031d01" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: WRITE    000305010000000302000000000078" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: WRITE    00040300010100" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: WRITE    00040900010000" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00" 
time="2017-02-28T19:24:04-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b03" 
time="2017-02-28T19:24:04-05:00" level=debug msg="transmitting frame: 200192010300000b003b03ecba" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b03000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:04-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b02" 
time="2017-02-28T19:24:04-05:00" level=debug msg="transmitting frame: 200192010300000b003b022d7a" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b020000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: WRITE    00040b3f800000" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: WRITE    00030701000000" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: READ     000104" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 0001045641524941424c45205350454544204655524e4143452020000000000000000000000000000000000000000000000000434553523133313437362d323420202035394d4e37413132305632343132323200000000313432354e2d2d2d2d2d2d20323631344134383935330000000000000000000000000000" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 5001: READ     000104" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 5001 -> 2001: RESPONSE 000104455645524553542054574f204341504143495459204f4455000000000000000000000000000000000000000000000000434553523133313339302d31312020003234414e42313438413030333130333000000000323431354e34313436333920323831354531333533370000000000000000000000000000" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: READ     000306" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 000306010254030202a8083e08" 
time="2017-02-28T19:24:04-05:00" level=debug msg="blower RPM is: 596" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 2001 -> 4001: READ     00030a" 
time="2017-02-28T19:24:04-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 00030a0175750300510075000010100114" 
[... 27 lines cut ...]
time="2017-02-28T19:24:06-05:00" level=debug msg="transmitting frame: 200192010300000b003b022d7a" 
time="2017-02-28T19:24:06-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b020000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:07-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b03" 
time="2017-02-28T19:24:07-05:00" level=debug msg="transmitting frame: 200192010300000b003b03ecba" 
time="2017-02-28T19:24:07-05:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b03000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000" 
time="2017-02-28T19:24:07-05:00" level=info msg="encoded frame: 9201 -> 2001: READ     003b02" 
[... cut ...]
time="2017-02-28T19:24:07-05:00" level=info msg="read frame: 5001 -> 2001: RESPONSE 003e0103cf03c803ff03ff000001013c" 
time="2017-02-28T19:24:07-05:00" level=info msg="heat pump coil temp is: 60.500000" 
time="2017-02-28T19:24:07-05:00" level=info msg="heat pump outside temp is: 60.937500" 
time="2017-02-28T19:24:07-05:00" level=info msg="read frame: 2001 -> 5001: READ     000311" 
time="2017-02-28T19:24:07-05:00" level=info msg="read frame: 5001 -> 2001: RESPONSE 00031125000000260000002a0002fc290000cc3d0000002c003617" 
pi@raspberrypi:~/infinitive$ 
```

acd, 2017-03-01:
Do you have a touchscreen thermostat?

ehwest, 2017-03-01:
yes, touchscreen

acd, 2017-03-04:
The same behavior was reported as issue #5.  I don't have a touchscreen thermostat or a SAM, so I'm unable to gather a protocol dump to try and reverse engineer the differences between non-touch and touch thermostats.

closed 2017-03-04 by acd

bvarner, 2017-03-04:
Can you post a dump of a non-touch thermostat?

I don't have the SAM, but I might be able to guess a bunch of things, and I'd be interesting in seeing what a non-touch thermostat sends.

ehwest, 2017-03-04:
I'm going to share this description of the protocol.  Its in RFT format.
The notion that structs would work on the raw data is suprising.  That approach relies on certain assumptions about how the compiler would interpret the raw data.  I guess if it works in some places, it should work everywhere.  It would break if they changed the number from a float to an integer.

[carrier_code_table.docx](https://github.com/acd/infinitive/files/818909/carrier_code_table.docx)