acd/infinitive #9: Bogus data

Will1604, 2017-03-14:
From the readme "Bogus data: Occasionally Infinitive will display incorrect data via the web interface for a second. This is likely caused by improper parsing of data received from the ABCD bus. I'd like to track down the root cause of this issue and resolve it, but due to its transient nature it's not a high priority and does not affect usability."

While this is transient on the web interface it's more of an issue when reading and logging data via the API. Previously when logging data every minute bogus data was showing up two or three times a day. Recently it's gone up to as much as 12 times a day. Still only 0.83% but it makes plots like the following pretty ugly
[image]

.

Will1604, 2017-03-14:
Created a branch to do two things
1. re request the table if the data looks bad
2. log the data received.

At the moment the determination of bad data is primitive but is covering the most common situations I've seen. A bad 003B03 register (TStatZoneParams) is consistently showing a HeatSetpoint and CoolSetpoint of 0°F. A bad 003B02 register (TStatCurrentParams) shows a current temp of 1°F. That's going to  need some improvement - though based on other observations I'm not sure that's going to be necessary.

What's odd is that looking in the log none of the data in the responses seems to be bad. It's just when the data is put into the structs that the problem occurs. 

Here's an example of a 003B02 request and response with the logging I've added. Note there is other activity on the bus interleaved with the requests by infinitive and the responses by the SAM. 

[003B02example.txt](https://github.com/acd/infinitive/files/840137/003B02example.txt)

All the visible responses by the thermostat back to Infinitive look fine - 0x46 corresponding to a current temperature of 70°F. The dump of the struct is definitely off though.

Will1604, 2017-03-14:
Here's an example from the log where bogus data is read for TStatZoneParams. requests and responses for the registers from the thermostat seem to get interleaved, but infinitive recognizes when it gets something it's not looking for. Nevertheless - a bogus TStatZoneParams gets through - the whole struct seems to be zeroed out i.e. it's probably just the initialized struct.

[003B03example.txt](https://github.com/acd/infinitive/files/841146/003B03example.txt)

Wondering whether a delay between the two requests from infinitive to the thermostat might keep things cleaner.

Will1604, 2017-03-16:
Added some additional logging (see [this commit](https://github.com/Will1604/infinitive/commit/08764b0b3eec23c6997471ba8fb1eee07c489273)) including the following line inside `func (p *InfinityProtocol) send`

`log.Debugf("Reading %v :%x", reflect.TypeOf(response), act.responseFrame.data)`

This showed up in the log. The valid response from the thermostat was on the bus (lines 1 and 2), for some reason the code is reading the wrong register into the TStatCurrentParams struct (line 4). It seems likely that the data actually being read is from the exchange between the thermostat heat pump in line 3.

```
time="2017-03-15T20:17:09-04:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b0201000045ffffffffffffff1818181818181818001e000000000304bf01"

time="2017-03-15T20:17:09-04:00" level=info msg="read frame: 2001 -> 9201: RESPONSE 003b0201000045ffffffffffffff1818181818181818001e000000000304bf01"

time="2017-03-15T20:17:09-04:00" level=info msg="read frame: 2001 -> 5001: READ    003e01"

time="2017-03-15T20:17:09-04:00" level=debug msg="Reading *main.TStatCurrentParams :003e01e84d2001500110000006003e0101da01d403ff03ff000000013c6a2192"

time="2017-03-15T20:17:09-04:00" level=debug msg="bad data obtained for TStatCur
rentParams"
```

Will1604, 2017-03-16:
The problem with TStatZoneParams seems to be different. Here's an example

[example9.txt](https://github.com/acd/infinitive/files/846434/example9.txt)

Here the first three bytes indicting the register (table row) are correct 003b03 its the following data that's not correct. For one thing it's too short. The dump of the struct shows the value planted in the Z8FanMode field at initialization is intact. It looks like nothing got copied into the struct. Probably binary.Read terminated early. Checking the return value from that function could be used to catch this. That wouldn't cover the situation where the bogus data were longer that the struct into which it is being read. Comparing the size of the data with the size of the struct is probably better.

Will1604, 2017-03-18:
Committed code to catch the two problems described above which result in bogus data.
Also added reporting to monitor how often those two problems are occurring.
See the [catchBadData](https://github.com/Will1604/infinitive/tree/catchBadData) branch.
Looking to see whether the work I have anything else ready to include in a pull request.

ehwest, 2017-03-23:
Carrier released and installed a new version 14.01 of their Touch Thermostat software.
The same zero-filled responses are apparent in the 003b02 and 003b03 messages.  I was told by the dealer that this new version enables Amazon Echo integration.   However, I don't see anything new here.  Maybe a search of all of the other tables/ rows would yield something new.

acd, 2017-03-23:
Thanks for looking in to this Will.  I've been rather busy but hope to have time to take a look at what you've done over the weekend.

Will1604, 2017-03-25:
Thanks Andrew. The comments above include my wandering towards a solution, so I'll try and summarize here what I think is worth looking at.

The changes, along with code to read the temperature unit, have been merged back into the Will1604/infinitive master branch . 

There's one thing in there that it would be good to get your take on. The following if statement in send() checks a number of conditions before processing a read. 
```
if ok && op == opREAD && act.responseFrame != nil && act.responseFrame.data != nil && len(act.responseFrame.data) > 6
```
If any of these is not met it seems to me the send() function should return false, but I could be missing the point so I didn't change it.

Root cause
The checks I've added for the register type and for the return value of binary.read catch where there is a problem. That seems good enough for my purposes but still doesn't solve the underlying cause. From the logs it looks like the memory for a given frame is getting overwritten before send() is done with it. Perhaps this is because of the way things are threaded, but I haven't explored that. You probably have a better understanding of that.
 
Error rate
I was seeing up to 0.8% bad data coming back through the API. With these changes the rate now seems to be about 0.03%, but all of that is trapped.

Despite all that; I glanced at the ui yesterday and bogus values flashed up and disappeared. Something is still getting through - but the bogus data appearing in my graphs has essentially disappeared.

acd, 2023-11-17:
I believe this is fixed in HEAD.

closed 2023-11-17 by acd