acd/infinitive #10: Heat stage messages are specific to electric heat system

Will1604, 2017-03-28:
My code to determine heat pump stage and whether electric heat was active was merged into main a couple of months ago. It wasn't as general as it might have been. The text in the UI and API was specific to a fan coil with auxiliary electric heat. For example in the UI it reports "Electric Heat Active". That's misleading for a system where the indoor unit is a furnace. 

The addFurnace branch sets out to provide text appropriate depending on whether the system has a hybrid heat furnace or a fan coil with electric heat.

I'd appreciate any preferences on what the API response should be

This covers all the bases but for any given system includes an element that is unnecessary.
`{"blowerRPM":1000, "airFlowCFM":1200, "elecHeat":true, "furnaceHeat":false}`

This would be more specific
A hybrid heat system would return
`{ "blowerRPM": 1000, "airFlowCFM":1200, "furnaceHeat":true }`
A fan coil/electric heat system would return
`{ "blowerRPM":0, "airFlowCFM":0, "elecHeat":false } `

Also; as I don't have a hybrid heat furnace I can't test whether the same bits are being used to indicate furnace heat, so for now I've included `"heatBits":"00000011"` in the response.

The branch also includes an API call to return details about the kind of units installed on the system.

Will merge this into will1604/master when done.

cheitzig, 2018-02-10:
I get a message (e.g., right now) that says HP Stage 1. I sort of assume this is heat pump related, yet I don't have a heat pump. I'm thinking your messages would fix that. I'd be happy to test / work with you to match your messages to what's going on. Tell me what next steps would be?

Will1604, 2018-02-11:
OK - I looked at my code yesterday. Am sure it's modifiable by adding different message strings. I've got some code I've not committed yet which calls each device to get its type. I won't be able to get to this until next weekend though.

cheitzig, 2018-02-12:
No hurry. I've just started to play with this software.

Will1604, 2018-02-18:
Checked in code to will1604/master with a simple fix to make the messages about heat stage generic for heatpump/furnace.

I don't yet feel in complete control of the process to incorporate changes in assets/ui.html to bindata_assetfs.go. So if the old (heat pump) messages are still showing then you can regenerate the binary version of the assets folder yourself 
go-bindata-assetfs assets/...

Packaged this change along with some other changes I had in progress
1. Shortened api URLs for devices which are not specific to a zone (heatpump, air handler)
2. Added api call to list the devices on the system
3. Added hold duration and humidity target to zone response
See the readme for the revised API calls and responses

cheitzig, 2018-02-19:
Thanks. I need to trouble-shoot an issue with my raspberry pi dropping off the network every 12-36 hours first, and then I can try your version. Thanks!