acd/infinitive #15: One small issue-- temp display at 1° F shows 255°

cheitzig, 2018-02-05:
See picture below. The current temp displayed on my thermostat is 1° F, but the Infinitive webpage shows 255°

[image]

cheitzig, 2018-02-06:
Update: at -2° F, the temp shows as 254° F, so it clearly seems to be a signed/unsigned 8-bit int issue. I haven't read through / understood all of your code, but wondering if this is as simple as changing OutdoorAirTemp (in tables.go, line 26) and OutdoorTemp (in infinitive.go, line 17) from an uint8 to an int8?

I'll try this, probably tonight, but I need to get a Go environment setup first.

cheitzig, 2018-02-07:
Confirmed that it seems to compile and work properly changing these two declarations to int8 instead of uint8. It's not below zero though, so I can't confirm. It's supposed to be below zero again on Friday.

Note: I don't know git well enough to understand how to propose changes. I believe I need to fork, make the change, and then propose a merge to you-- is that correct?

rcojoe, 2018-02-07:
Cheitzig, Mine shows 255 too.  I don't have an outside temp probe set up and my thermostat doesn't show an outside temperature for that reason.   Just giving you my feedback on what I am seeing.

cheitzig, 2018-02-10:
Confirmed today that my fix fixes the negative temperature issue. Slightly amusing backstory though: I was hoping it was going to be < 0° F, and even got up fairly early, but my thermostat showed 0° F when I got up. But it occurred to me to switch the thermostat into metric mode and then we'd be solidly below zero. Tried it both ways and confirmed my proposed fix solves the problem.

Below is temperature display before the change of the uint8 to int8 in two places in the code:
[image]

Below is temperature display after the change of the uint8 to int8 in two places in the code:
[image]

rcojoe, 2018-02-12:
@cheitzig Have you seen the work that Will1604 has done?  https://github.com/Will1604/infinitive/network

I was reviewing some of his code, and it seems he has taken the interface and API a bit further. The ACD repository doesn't have all of his changes committed yet.

cheitzig, 2018-02-12:
Thanks. I exchanged comments with @Will1604 about his heat pump messages post, so I'll maybe talk to him next weekend when he has some time.

closed 2023-11-17 by acd