acd/infinitive #17: What is a SAM?

Pascal666, 2018-06-01:
It would be nice if README.md (the file shown on the GitHub project page) started with a paragraph explaining what this software does.  Just saying it "impersonates a SAM" is useless if the reader has no idea what a SAM is (like me).  See the README.md for Infinitude for a good example.

I'm looking for some way to control my Bryant condenser via the ABCD bus, without buying a bunch of expensive hardware, because the Y1/Y2 terminals were damaged by lightning.  I don't have any other Bryant equipment.  After reading the README.md I still have no idea what this software does or if it will do what I need.

Thank you.

Will1604, 2018-06-06:
SAM stands for a System Access Module. It provides a serial (RS-232) interface to allow a home automation system to read from and send commands to your HVAC system. It also provides for remote access to your HVAC system. One version of SAM is described [here](https://www.carrierenterprise.com/carrier-infinity-series-ethernet-cat-5-wired-broadband-remote-access-module-systxccrct01) . 

There's a document [here](http://dms.hvacpartners.com/docs/1009/Public/02/SAM01-01XA.pdf) which describes the connection and the protocol for communicating with the SAM which gives some sense of what it can do.

Infinitive impersonates a SAM in the general sense of being able to read data from the HVAC system and send commands to it. It does not use the same commands or syntax. Communication with the system is via a REST interface described in the Infinitive readme file.

Infinitive also provides a web interface to display and control at least some of the aspects of the HVAC system from a web browser on a computer, phone, tablet. etc. As it stands Infinitive does not provide remote access other than via any general mechanism that makes a computer on your local network accessible on the internet or a VPN.

SAM's no longer feature prominently in Bryant/Carrier's marketing material if at all. The capabilities have largely been superseded by smart thermostats e.g. [this one](https://www.bryant.com/bryant/en/us/products/controls-thermostats/systxbbecc01-b/) in Bryant's case. These are probably the best comparison for Infinitive as of now.

Will1604, 2018-06-07:
...and regarding controlling the condenser. Infinitive sends its commands to the thermostat on the bus which in turn communicates with the other devices on the bus. It may be possible to emulate the commands from the thermostat to the condenser but you would have to change the code to do so.

Pascal666, 2018-06-07:
Thank you for the excellent explanation.  I hope that ends up in README.md someday.