acd/infinitive #18: Hardware connection options

pfiggins, 2018-06-06:
Anyone know what locations are possible for connecting the hardware? I see mention of connecting wires to the air handler. But in my case, the thermostat location would be far easier to access. I'm not familiar with the rs485 communication protocol at all. Is it possible to piggyback my pi to the terminals of the Carrier thermostat? Or would I have to remove the wires from the existing thermostat and then connect them to the pi? If connectign at the air handler, does that require removing the existing wires, or just add an additional set of wires going to the pi? Ideally, I leave the existing Carrier thermostat in place, just for ease of use.

bvarner, 2018-06-06:
You should be able to connect at any point, so long as you wire in parallel.

If you're using an off-the-shelf RS485-USB (or TTL) device, then you're probably good to go on termination, isolation, bias, and shouldn't need to worry about it.

So, If you leave the thermostat in place, you should be able to add some jumpers to the screw-terminals on the back of the thermostat base plate to make your connection.

pfiggins, 2018-06-06:
Sweet. I actually found a guy who makes a 24V + RS485 Pi HAT. So my RS485 isn't exactly off the shelf. But if it winds up working, it will make a heck of a setup for this use case! He did offer to help me get it working if I had issues implementing, so hopefully we can make it happen.

bvarner, 2018-06-07:
If it’s the same thing I saw on eBay you may want to be careful with that. 

The carrier systems are 24v AC!

pfiggins, 2018-06-07:
Thank you for catching that. Not sure how I missed it. Wow, that would have been bad! 

Looks like back to the drawing board. Such a shame because I don't think I have easy access to power elsewhere, so I'm going to have to figure something out for powering the pi now.

gozoinks, 2018-06-07:
Bridge rectifier?

pfiggins, 2018-06-07:
Thanks, guys! I'm a software guy, not a hardware guy. So I'm out of my element here. Thinking this should do the trick - https://smile.amazon.com/gp/aw/d/B00RE6QN4U
Input range: DC 7v~52v /AC 8V - 52V 
Output voltage:5V 
Output current/power:3A 
Efficiency: ＞96% 

Should be able to pair it with one of these FTDI based RS485 adapters and get the project going!

pfiggins, 2018-11-15:
I finally had a chance to work on this project and I wired up that 5V/3A converter linked above to power the pi from the 24V AC at the thermostat (wires C and D). It powered the pi fine and ran without saying it was throttled. But almost every message in the rs485 communications was a timeout. I tried running my rs485 ground to a ground on the gpio, just in case. No luck there. I also tried tying that ground with the ground on my power adapter too. Still no luck. I tested with the multimeter and it's giving a solid 5.01V DC from the 27.3V AC source, but I guess it must not be putting out enough amps to properly power the rs485 adapter over the USB bus? I plugged a normal power brick into the wall and tried again, and it all worked fine! 

Anyone have any recommendations for an adapter that is up to the task? I am hoping to piggyback on the 24V because I'm mounting the pi in the spot for our old humidistat. If I can't use the 24V, I'll have to pull wires and mount an electric box to get this thing powered. That seems lame considering I have power right there, just not the right voltage...

nebulous, 2018-11-15:
Did you check with a multimeter when the pi was powered up and under load? I'm certain any buck converter could output a steady 5vDC from the 24vAC supply, but a variable load changes matters.
*hopefully you can get this to work. I've been toying with the idea of running an esp8266 with a max485 from the 24vAC wires for someTimeNow™

pfiggins, 2018-11-15:
Yep, I checked it across the pads on the back of the pi while it was running and the infinitive script was hitting the rs485 adapter... PP1 and PP3 I think it was? It was reading a rock solid 5.01v.

nebulous, 2018-11-15:
Buck converters are switching supplies, so problems may simply be noise. You could try a ferrite suppressor around the input and/or a capacitor across power/gnd to smooth things out a bit.

pfiggins, 2018-11-15:
Thanks for the idea. I'm way out of my element when it comes to power supplies. Did some googling, but I'm not sure I found any real answers to this so I came back here... Would you have a suggestion of what size cap to put across 5v and gnd? And would I put it across particular pins on the gpio? Or just any 5v and gnd? I'm sure I have some caps around here, not so much with the ferrite suppressor.

Sorry if that seems like a stupid question. Just a bit clueless at this level of hardware.

bvarner, 2018-11-15:
From what I recall with the ABCD lines, the 24vAC has a fairly low forward current before you'll fry fuses inside your unit. I thought I'd heard something about zoned systems only moving one damper at a time to compensate for current demands and keep from overloading the fuses.

I've not found (or made) a 24vac -> 5v regulated supply for my setup (yet), but probably ought out try it out.

I saw someone mention tying an esp8266 to a max485... having worked directly with the max485 on my own version of a TTL->RS485 converter -- If you're planning on 'writing' to the bus instead of just reading / snooping it, it's not as straightfoward as you may want to think. You'll also need to account for power suppression, surges, incorrect connections in other parts of the network, signal line biasing, etc... if you want it to be a 'robust' 485 connection. There's an art to it, and by the time I finally figured out all those things, I found a really inexpensive TTL->RS485 adapter board with all the bells and whistles that didn't require me to engineer it on my own, and it was _super_inexpensive_.

Fully isolating adapters which properly bias and protect the signal lines, properly handle writing (or, latching the write enable pin of a max485 based on input signals) to the bus, etc. run about $0.70 from  aliexpress suppliers. Don't get one that isn't isolating. It's not worth blowing up your HVAC board.

pfiggins, 2018-12-01:
Thanks for the helpful info, guys! I'm back from vacation and hoping to get back on this project. I'm still a little clueless as to the proper approach to fixing my power issue here.

@nebulous mentioned a ferrite suppressor or cap. I can snag the ferrrite suppressor on amazon and give it a shot. You say to put it on the input side. We talking about the data wires coming from the ABCD bus to my RS485 adapter? And if I were to try the capacitor route, what size capacitor would I want to use? Does it matter which 5v/gnd pins I put the cap across?

@bvarner mentioned a low forward current. I have no idea what forward current means, but I'm guessing the fuses in the system are low amperage and we risk blowing them if we draw too much? Is there a way to determine how much current I can safely pull? Looking at a a fuse size somewhere in the air handler maybe? That said, nothing has blown on me so far. And it is powering the pi... Just not cannot seem to also properly power the RS485 USB adapter. Or at least that is my best guess as to why the RS485 communication is failing when running off this power adapter vs the wall wart. Might @nebulous suggestion of using a capacitor possibly cure this issue too? In that if we have power demand spikes, the cap could supply them rather than drawing from the actual 24v supply line?

bvarner, 2019-01-08:
Yep, I think they're standard fuses in the hvac system -- but I'd never change them with larger fuses...
It sounds like your converter may not be able to supply enough amps -- either as a consequence of step-down efficiency, or just as a consequence of the nature of things.

I've had issues with pi's and USB devices sucking too much power for low amperage phone chargers. I had one old keyboard/mouse combo that would draw enough power that the USB bus would just start dropping devices on an original model B.

mangusman07, 2020-01-26:
> But almost every message in the rs485 communications was a timeout. 

@pfiggins do your timeouts look similar to the logs in https://github.com/acd/infinitive/issues/20

I'm just getting started with troubleshooting infinitive, and would wager your problems are tied to software not wiring. Do you have an older thermostat?