acd/infinitive #19: RS-485 seems OK but UI is all Zero values!

pcornill, 2018-06-08:
I have set up the RPi with a RS-485 to USB FTDI adaptor. I can see the messages like below.
Unfortunately, running le Web interface or API doesn’t work for me. I have the following result!
Did I miss something ? I just run:
./infinitive -httpport=8080 -serial=/dev/ttyUSB0

INFO[0000] read frame: 5001 -> 2001: RESPONSE 000104455645524553542054574f204341504143495459204f4455000000000000000000000000000000000000000000000000434553523133313339302d31312020003234414e42373336413030333130333000000000323531344e34323238323120323831344532363433330000000000000000000000000000
INFO[0000] read frame: 4001 -> 2001: RESPONSE 0003060100000000000005aa00
DEBU[0000] blower RPM is: 0
INFO[0000] read frame: 4001 -> 2001: RESPONSE 00030a017272030051004e000010080114
INFO[0000] read frame: 5001 -> 2001: RESPONSE 003e0103ac039303ff03ff000001013c
INFO[0000] heat pump coil temp is: 57.187500
INFO[0000] heat pump outside temp is: 58.750000

marcusadolfsson, 2018-09-17:
I am having a similar issue with Bryant Evolution setup. I can see the heat pump temps in the logs but nothing else (no blower RPM etc). With infinitude I can see the data on their serial page so I know the interface is working correctly.

marcusadolfsson, 2018-09-17:
Update looks like the code is parsing the air handler and heatpump fine, but not the thermostat.

/api/zone/1/airhandler
{"blowerRPM":1223,"airFlowCFM":908,"elecHeat":false}

/api/zone/1/heatpump
{"coilTemp":99.3125,"outsideTemp":87.6875,"stage":2}

/api/zone/1/config
{"currentTemp":0,"currentHumidity":0,"outdoorTemp":0,"mode":"heat","stage":0,"fanMode":"auto","hold":false,"heatSetpoint":0,"coolSetpoint":0,"rawMode":0}

marcusadolfsson, 2018-09-17:
Think I am getting closer. Here is the thermostat responding to the fake SAM, but something is wrong.

INFO[1169] read frame: 2001 -> 9201: RESPONSE 003b03000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 
INFO[1169] got response for incorrect table, is: 012001 expected: 003b03

bdf0506, 2019-03-06:
Hi, any updates on this issue @marcusadolfsson ?

I just tried to install this and have a similar issue. My screenshot has all 0s like the one that @pcornill shows, except mine does in fact show the blower RPM.

When I look at logs I see the following:

```
Mar  6 14:46:48 server infinitive[929]: time="2019-03-06T14:46:48-05:00" level=info msg="read frame: 2001 -> 4001: READ     000104"
Mar  6 14:46:48 server infinitive[929]: time="2019-03-06T14:46:48-05:00" level=info msg="read frame: 4001 -> 2001: RESPONSE 0001045641524941424c45205350454544204655524e4143452020000000000000000000000000000000000000000000000000434553523133313331322d313520202035384356413039302d2d2d313531313600000000333131304e33313639333420333331304131363237370000000000000000000000000000"
Mar  6 14:46:48 server infinitive[929]: time="2019-03-06T14:46:48-05:00" level=info msg="Error reading device info|EOF|"
```

And then when I try to pull it up the web UI, I see the following:

```
Mar  6 14:47:06 server infinitive[929]: time="2019-03-06T14:47:06-05:00" level=debug msg="heat pump coil temp is: 42.125000 F"
Mar  6 14:47:06 server infinitive[929]: time="2019-03-06T14:47:06-05:00" level=debug msg="heat pump outside temp is: 49.375000 F"
Mar  6 14:47:06 server infinitive[929]: time="2019-03-06T14:47:06-05:00" level=info msg="error on websocket write: write tcp 192.168.30.215:8080->192.168.20.101:61818: write: broken pipe"
Mar  6 14:47:06 server infinitive[929]: time="2019-03-06T14:47:06-05:00" level=info msg="closing websocket"
Mar  6 14:47:06 server infinitive[929]: time="2019-03-06T14:47:06-05:00" level=info msg="error on ws close: write tcp 192.168.30.215:8080->192.168.20.101:61818: write: broken pipe"
```

bdf0506, 2019-03-07:
Looking into this more, seems like this is behavior on a touch thermostat. You have to use a project like infinitude instead and proxy the connection through there, as things like temperature live on the touch thermostat and not the ABCD bus.

nebulous, 2019-03-07:
It's entirely possible(and I think probable since touch stats can be made aware of SAMs) that settings can be pulled from a touch stat, but at minimum they live at different - as yet undetermined - registers and would need to be queried because the stat has no need to advertise them to other devices.

gartz, 2021-02-05:
Interesting discussion, I'm having the same issue here using the touch thermostat. What intrigues me is that external sensors like the outside temperature is advertised using the BUS to the thermostat, but it's not interpreted by infinitive.

mattster98, 2021-06-11:
This code advertises as a SAM though, right?  Does that mean that even a SAM wouldn't have access to this data?

mattster98, 2021-06-11:
@marcusadolfsson did you ever make any further progress here?  I just added a new system and figured I'd try this again and so far am only parsing air handler fan speed.  /api/devices returns a blank page which I suspect is a bad sign.  Lots of data, but not getting parsed into anything useful.

ericreich, 2023-01-20:
Any updates to this....

I am only seeing FAN motor rpm and CFM. So I know I am connected and working correctly. Everything else are zeros for temps etc.. I am guessing infinitive is not aware of this variable speed furnace, tstat, and variable speed condenser? Any help would be great.

Eric

bokaklav, 2025-07-15:
> I am only seeing FAN motor rpm and CFM
@bdf0506 @ericreich  I'm getting this behavior.

I'm running this on an early 2000s era Carrier Inifinity Model FE4ANF003.   I had to build from the master branch to even be able to see the blower rpms.  Below is my output

`DEBU[1024] timeout waiting for response, retransmitting frame 
DEBU[1024] transmitting frame: 200192010300000b003b03ecba
INFO[1024] read frame: 2001 -> 4201: READ     000306    
INFO[1024] read frame: 4201 -> 2001: ACK06    000306010497000000c8067208 
DEBU[1024] blower RPM is: 1175
INFO[1025] read frame: 2001 -> 9201: NACK     04        
INFO[1025] read frame: 2001 -> 4201: READ     000402    
INFO[1025] read frame: 4201 -> 2001: ACK06    0004021d3537ca2a1d34e95b371db4f8b0791d0000000002d502d502d5 
DEBU[1025] timeout waiting for response, retransmitting frame 
DEBU[1025] transmitting frame: 200192010300000b003b03ecba
INFO[1025] action timed out`

`curl http://:8080/api/airhandler
{"blowerRPM":1163,"airFlowCFM":1075,"elecHeat":false}

curl http://:8080/api/zone/1/heatpump
{"tempUnit":"","coilTemp":0,"outsideTemp":0,"stage":0}`