acd/infinitive #20: RS-485 Errors and Timeouts

madsensh, 2018-06-19:
Can I get a little help troubleshooting my setup?

I have a Carrier Infinity furnace and AC unit that were installed in 2005.  When I moved in about 3 years ago it was connected to a generic 2-stage thermostat, which I replaced with a systxccuid01 thermostat (the one with buttons) from eBay along with a new run of 18/5 thermostat wire from the furnace in the attic to the AC unit outside (about 50 feet).  The thermostat is directly below the furnace, connected by about 10 feet of wire.  Over the weekend, I ran about 20 feet of new 18/2 thermostat wire from the furnace to an RPi in my structured media center, which has the following RS-485 to USB adapter: [https://www.amazon.com/gp/product/B005CPI0JQ ](url)

The Infinity A and B wires are connected to the RS-485 (A+) and (B-) terminals, respectively, which I believe is correct, although I'm new to RS-485 and it feels weird to not have a signal ground wire like with RS-232.

When I fire up Infinitive, I see lots of errors in the log output (see below), and the Infinitive UI is not populated with any values.  (note that the AC was not on at the time, so RPM and CFM readings of 0 are correct)

Any ideas where I should direct my troubleshooting efforts?  Does this look like a wiring issue, or possibly something wrong on the Carrier side?  I'd rather not put the RPi at the furnace permanently, as I have a very hot attic, but maybe I can try it temporarily to rule out wiring issues due to staples, running near AC circuits, etc.

INFO[0028] read frame: 2001 -> e01: READ     003e04
INFO[0028] read frame: e01 -> 2001: RESPONSE 003e0400
INFO[0028] read frame: 2001 -> e01: READ     003e07
INFO[0028] read frame: e01 -> 2001: RESPONSE 003e070000
INFO[0028] read frame: 2001 -> 9201: ERROR    04
DEBU[0028] timeout waiting for response, retransmitting frame
DEBU[0028] transmitting frame: 200192010300000b003b03ecba
INFO[0028] read frame: 2001 -> 9201: ERROR    04
DEBU[0028] timeout waiting for response, retransmitting frame
DEBU[0028] transmitting frame: 200192010300000b003b03ecba
INFO[0028] read frame: 2001 -> 9201: ERROR    04
DEBU[0028] timeout waiting for response, retransmitting frame
DEBU[0028] transmitting frame: 200192010300000b003b03ecba
INFO[0028] read frame: 2001 -> 9201: ERROR    04
DEBU[0028] timeout waiting for response, retransmitting frame
DEBU[0028] transmitting frame: 200192010300000b003b03ecba
INFO[0028] read frame: 2001 -> 9201: ERROR    04
DEBU[0029] timeout waiting for response, retransmitting frame
DEBU[0029] transmitting frame: 200192010300000b003b03ecba
INFO[0029] action timed out
INFO[0029] read frame: 2001 -> 9201: ERROR    04
INFO[0030] encoded frame: 9201 -> 2001: READ     003b03
DEBU[0030] transmitting frame: 200192010300000b003b03ecba
INFO[0030] read frame: 2001 -> 9201: ERROR    04
INFO[0030] read frame: 2001 -> 4001: READ     00030a
INFO[0030] read frame: 4001 -> 2001: RESPONSE 00030a010a0a02001f004b000010080414
INFO[0030] read frame: 2001 -> 4001: READ     000104
DEBU[0030] timeout waiting for response, retransmitting frame
DEBU[0030] transmitting frame: 200192010300000b003b03ecba
INFO[0030] read frame: 4001 -> 2001: RESPONSE 0001045641524941424c45205350454544204655524e4143452020000000000000000000000000000000000000000000000000434553523133313331322d303620202035384d56503038302d462d31383131340000002d343330344e3030313935302034353034413130333634000000000000000000000000002d
INFO[0030] read frame: 2001 -> 9201: ERROR    04
INFO[0030] read frame: 2001 -> 4001: READ     000306
INFO[0030] read frame: 4001 -> 2001: RESPONSE 000306010000000001e505aa00
DEBU[0030] blower RPM is: 0
INFO[0030] read frame: 2001 -> 4001: READ     000302
INFO[0030] read frame: 4001 -> 2001: RESPONSE 000302041100000414000004020000
INFO[0030] read frame: 2001 -> 4001: READ     000316
DEBU[0030] timeout waiting for response, retransmitting frame
DEBU[0030] transmitting frame: 200192010300000b003b03ecba
DEBU[0030] timeout waiting for response, retransmitting frame
DEBU[0030] transmitting frame: 200192010300000b003b03ecba
DEBU[0030] timeout waiting for response, retransmitting frame
DEBU[0030] transmitting frame: 200192010300000b003b03ecba
DEBU[0031] timeout waiting for response, retransmitting frame
DEBU[0031] transmitting frame: 200192010300000b003b03ecba
INFO[0031] action timed out
INFO[0031] read frame: 2001 -> 4001: READ     000316
INFO[0031] read frame: 4001 -> 2001: RESPONSE 0003160000000000000000005a01000352
DEBU[0031] air flow CFM is: 0
INFO[0031] read frame: 2001 -> 9201: ERROR    04
INFO[0031] read frame: 2001 -> 9201: ERROR    04
INFO[0031] read frame: 2001 -> 9201: ERROR    04
INFO[0031] read frame: 2001 -> 4001: READ     000402
INFO[0031] read frame: 4001 -> 2001: RESPONSE 0004025a7896b45a1d358ec4741d34978da41db4eaad511d324813a602d005aa

madsensh, 2018-06-22:
Quick update... I was reading through one of the old [threads] (http://cocoontech.com/forums/topic/11372-carrier-infinity/page-8) that goes into more detail on the process folks have followed to reverse engineer the RS-485 protocol.  Based on that work, I believe my "ERROR 4" is a "no table" error, most likely due to my having an older single-zone thermostat, whereas infinitive may have been tested using newer software releases and/or multi-zone stats.  Next step will be to log some traffic and map out my stat's table structure.

ratatine, 2019-08-15:
It seems I'm in the same boat as madsensh. Anyone ever make progress on this issue from last year?

In my case my system is also from about 2005. It has three zones with a damper control module, a CESR131339-06 thermostat and two remote thermostats.

A log sample from mine is below. After letting infinitive run for a while it does pickup the blower speed but that's about it so far. No temp, API is blank or responded with 404s. I also see the e01 device from the looks of it.

INFO[0991] read frame: 2001 -> 6001: WRITE    00340400
INFO[0991] read frame: 6001 -> 2001: RESPONSE 00
INFO[0991] read frame: 2001 -> 6001: WRITE    00030806000f0000000000
INFO[0991] read frame: 6001 -> 2001: RESPONSE 00
INFO[0991] read frame: 2001 -> 6001: READ     003404
INFO[0991] read frame: 6001 -> 2001: RESPONSE 00340400
INFO[0991] read frame: 2001 -> 9201: ERROR    04
INFO[0991] read frame: 2001 -> 9201: ERROR    04
INFO[0991] read frame: 2001 -> 4001: READ     00030d
INFO[0991] read frame: 4001 -> 2001: RESPONSE 00030d00000000000000
INFO[0991] read frame: 4001 -> 4101: READ     00030a
INFO[0991] read frame: 2001 -> 9201: ERROR    04
INFO[0991] read frame: 4001 -> 4101: READ     00030a
INFO[0991] read frame: 4001 -> 4101: READ     00030a
INFO[0991] read frame: 4001 -> 4101: READ     00030a
INFO[0991] read frame: 4001 -> 4101: READ     00030a
DEBU[0991] timeout waiting for response, retransmitting frame
DEBU[0991] transmitting frame: 200192010300000b003b03ecba
INFO[0991] read frame: 4001 -> 4101: READ     00030a
DEBU[0991] timeout waiting for response, retransmitting frame
DEBU[0991] transmitting frame: 200192010300000b003b03ecba
INFO[0991] action timed out
INFO[0991] read frame: 2001 -> 9201: ERROR    04
INFO[0991] read frame: 2001 -> f1f1: WRITE    000420d00ac0000f0104e802012d3758003d4e00000000
INFO[0992] read frame: 2001 -> f1f1: WRITE    0002020c2a04
INFO[0992] read frame: 2001 -> f1f1: WRITE    000203100813
INFO[0992] encoded frame: 9201 -> 2001: READ     003b03
DEBU[0992] transmitting frame: 200192010300000b003b03ecba
INFO[0992] read frame: 2001 -> 9201: ERROR    04
DEBU[0992] timeout waiting for response, retransmitting frame
DEBU[0992] transmitting frame: 200192010300000b003b03ecba
INFO[0992] read frame: 2001 -> 9201: ERROR    04
DEBU[0993] timeout waiting for response, retransmitting frame
DEBU[0993] transmitting frame: 200192010300000b003b03ecba
INFO[0993] read frame: 2001 -> 9201: ERROR    04
INFO[0993] read frame: 2001 -> e01: READ     003e07
INFO[0993] read frame: e01 -> 2001: RESPONSE 003e070000
DEBU[0993] timeout waiting for response, retransmitting frame
DEBU[0993] transmitting frame: 200192010300000b003b03ecba
INFO[0993] read frame: 2001 -> 9201: ERROR    04
INFO[0993] read frame: 2001 -> 4001: READ     00030a
INFO[0993] read frame: 4001 -> 2001: RESPONSE 00030a010c0c02001f005d000010100414
INFO[0993] read frame: 2001 -> 4001: READ     000104
INFO[0993] read frame: 4001 -> 2001: RESPONSE 0001045641524941424c45205350454544204655524e4143452020000000000000000000000000000000000000000000000000434553523133313331322d303620202035384d56503130302d462d31383132300000002d333430344e3030323132332033373034413131353130000000000000000000000000002d
INFO[0993] read frame: 2001 -> 4001: READ     000306
DEBU[0993] timeout waiting for response, retransmitting frame
DEBU[0993] transmitting frame: 200192010300000b003b03ecba
DEBU[0993] timeout waiting for response, retransmitting frame
DEBU[0993] transmitting frame: 200192010300000b003b03ecba
INFO[0993] action timed out
INFO[0994] encoded frame: 9201 -> 2001: READ     003b03
DEBU[0994] transmitting frame: 200192010300000b003b03ecba
INFO[0994] read frame: 2001 -> 4001: READ     000306
INFO[0994] read frame: 4001 -> 2001: RESPONSE 0003060103ca057802a807da08
DEBU[0994] blower RPM is: 970
INFO[0994] read frame: 2001 -> 4001: READ     000302
INFO[0994] read frame: 4001 -> 2001: RESPONSE 000302041100000414000004020000
INFO[0994] read frame: 2001 -> 4001: READ     000316
INFO[0994] read frame: 4001 -> 2001: RESPONSE 00031600000200057800000078010002d0
DEBU[0994] air flow CFM is: 1400
INFO[0994] read frame: 2001 -> 9201: ERROR    04
INFO[0994] read frame: 2001 -> 4001: READ     000402
INFO[0994] read frame: 4001 -> 2001: RESPONSE 0004025a7896b4781d35fde6511db5c028d61d35503fe21db48bfa7503a205f0
INFO[0994] read frame: 2001 -> 4001: WRITE    00040300010100
[... 226 lines cut ...]
INFO[1006] read frame: 2001 -> 9201: ERROR    04
INFO[1006] read frame: 2001 -> f1f1: WRITE    000420d00ac0000f0104e802012d3758003d4e00000000
DEBU[1006] timeout waiting for response, retransmitting frame
DEBU[1006] transmitting frame: 200192010300000b003b03ecba
INFO[1007] read frame: 2001 -> 9201: ERROR    04
DEBU[1007] timeout waiting for response, retransmitting frame
DEBU[1007] transmitting frame: 200192010300000b003b03ecba
INFO[1007] read frame: 2001 -> 9201: ERROR    04
INFO[1007] read frame: 2001 -> 4001: READ     00030d
INFO[1007] read frame: 4001 -> 2001: RESPONSE 00030d00000000000000
DEBU[1007] timeout waiting for response, retransmitting frame
DEBU[1007] transmitting frame: 200192010300000b003b03ecba
INFO[1007] read frame: 2001 -> 9201: ERROR    04
DEBU[1007] timeout waiting for response, retransmitting frame
DEBU[1007] transmitting frame: 200192010300000b003b03ecba
INFO[1007] read frame: 2001 -> 9201: ERROR    04
DEBU[1007] timeout waiting for response, retransmitting frame
DEBU[1007] transmitting frame: 200192010300000b003b03ecba
INFO[1007] action timed out
[... cut ...]
INFO[1008] read frame: 2001 -> 4001: READ     00030a
INFO[1008] read frame: 4001 -> 2001: RESPONSE 00030a010c0c02001f005d000010100414
INFO[1008] read frame: 2001 -> 4001: READ     000104
INFO[1008] read frame: 4001 -> 2001: RESPONSE 0001045641524941424c45205350454544204655524e4143452020000000000000000000000000000000000000000000000000434553523133313331322d303620202035384d56503130302d462d31383132300000002d333430344e3030323132332033373034413131353130000000000000000000000000002d
INFO[1008] read frame: 2001 -> 4001: READ     000306

mangusman07, 2020-01-26:
Any updates / successes in the past few months?

I'm away from home for the weekend, but I just got infinitive running and see a very similar set of errors and can only read fan speed and CFM. Software is not my best strength, but I have a strong desire to get this running. I also have an older thermostat with one zone defined.

I currently have it wired with some simple 20 awg stranded wire, and have seen claims that people have trouble getting Infinitive/Infinitude running without solid core wire. For low current systems, wire gauge and stranding should have minimal effect on signal strength so I believe this is a software problem.

ratatine, 2020-01-26:
No progress on my side. It is very version specific from whst I've concluded so if I changed out my thermostat with the right version it would work. Alas that's not happening.

minideezel, 2020-02-03:
I am also having the same issues described in this thread my my single zone 2005 unit. It's a bummer because I was really excited to have a functioning connected thermostat for the cost of a RS485 usb adapter and raspberry pi! I'm ok at debugging things so I might try to look into it once I have time, but unlikely. Would be great if someone with knowledge in this program could provide some guidance.

hornsbydm, 2022-01-02:
I know this thread is a bit dated. I have a Bryant Evolution System with the CESR13339-16 thermostat similar to OPs. I can read data off the bus just fine however the thermostat will not respond to any requests from 9201. 

I believe this to be because infinitive fails to fully impersonate a SAM. Upon startup the thermostat polls all device types including 9201. For the SAM it is requesting to read registers 0x003f02, and 0x00040e. I think because it does not get a response to this polling it will drop any READS 9201 -> 2001.

Does anyone that has a dump from a real SAM know what's in these registers?