acd/infinitive #21: Airhandler page missing?

mww012, 2018-08-21:
First I'd like to say thanks for all the effort put into this project.  I bought a new house and afterward found I couldn't use any "smart" thermostat aside from ones from Bryant.  Those are outside my budget and this software has been perfect for my needs.

My issue is that the /api/zone/1/airhandler page doesn't seem to exist.  I'm running v0.2 on a RPi zero (no performance issues so far!).  Anytime I try to hit it I get a 404 in response.  The main UI is showing the blowerRPM info so I know infinitive is pulling the data.  /api/zone/1/config returns a 200 and a son response as expected.  Am I missing something simple?

Will1604, 2018-08-21:
If you downloaded the 0.2 release I'm not sure it includes the air handler and heat pump additions to the API. You would have to build from source to get that.

See also will1604/infinitive for some additional changes. I shortened the API URLs for air handler and heat pump calls as these are at the system level rather than within a zone. Also added a URL endpoint  to list the devices on the system.

mww012, 2018-08-26:
Thanks for the response!  Sorry it took me a week to reply.

I just got your Will1604/infinitive implementation built and it worked like a charm!  I'm working on building a python infinitive api wrapper and from there building a platform for the @home-assistant project.  I'm not a dev by any stretch of the imagination but I'm wanting to sharpen my skills.  Seems like a good place to start.

Thanks again for your help!

closed 2018-08-26 by mww012

Will1604, 2018-08-28:
Glad you got it working. I was planning to clean a couple of things up before putting a pull request to get these changes back into and/infinitive. Adding builds is also on my to do list.