acd/infinitive #25: Is setting target humidity possible?

mww012, 2019-06-13:
I've been running a custom component for the Home Assistant project that uses the infinitive API to create a "smart" HVAC system.  I had a user ask if it were possible to set the target humidity.  I started digging and I see references to setting target humidity in the code but it's not working for me.  Was this a feature that never was completed?

Will1604, 2019-06-14:
If we can work out how to set the flag that corresponds to target humidity then it seems doable. From the documentation of the Infinity protocols that's available I'm not sure the right value is known.

For example in web server.go this is how the cool point it set is as follows. In webserver.go
```
			if args.CoolSetpoint > 0 {
				params.Z1CoolSetpoint = args.CoolSetpoint
				flags |= 0x08
			}
```
We need something like:
```
			if args.HumiditySetpoint > 0 {
				params.Z1TargetHumidity = args.HumiditySetpoint
				flags |= // need the value for this to indicate humidity should be changed
			}
```

mww012, 2019-06-28:
Thanks for the explanation. I’m willing to monitor and pull logs if that’s what we need. I’d need a little guidance on exactly how to do that though.

closed 2020-03-18 by mww012