acd/infinitive #26: Airhandler/Heatpump API broken?

haberda123, 2019-12-15:
The Airhandler and Heatpump API does not work.  I am receiving a 404 when I request it.  Is this a known issue?  I am looking through the code and I think it may have something to do with getAirHandler(?) function in the infinitive.go file?

Will1604, 2019-12-17:
Have to get through obvious questions first.
Is the user interface working?
What URL are you using for the api?

If the ui is at
http://yourhost:8080 and http://yourhost:8080/ui/
then the heat pump and air handler would be at
http://yourhost:8080/api/heatpump
http://yourhost:8080/api/airhandler

haberda123, 2019-12-17:
Sorry for not being clear.  Infinitve has been working with my Bryant furnace for some time (over a year) and I utilize the API (config) with my automation system in my home.  The defined URL's for the API is as you mentioned above.  After posting this I looked in the closed issues section and I saw that the downloaded version does not include the airhandler/heatpump API and I have to build from source with go and I will try that this weekend... I suspect that it will work since it resolved others that have run into this.  I will report back once I can sit down and see if it resolves the issue.

Will1604, 2019-12-17:
I guess I could do a build on the Will1604 fork of Infinitive, but I wouldn't be able to get to that until the new year.

haberda123, 2019-12-20:
I did get compile and build from source and everything looks good as far as the API.  Not sure if we want to leave this issue open in case somebody else runs into this!

Thanks!

closed 2023-11-17 by acd

raygamero, 2024-06-11:
I'm having the same problem,  I installed this code on two PIs years ago. never been able to pull fan speed.  
C:\>curl -X GET http://192.168.4.159:8080/api/zone/1/config
{"currentTemp":74,"currentHumidity":51,"outdoorTemp":69,"mode":"auto","coolActive":false,"heatActive":false,"fanMode":"auto","hold":false,"heatSetpoint":71,"coolSetpoint":74}

C:\>curl -X GET http://192.168.4.159:8080/api/airhandler
404 page not found
C:\>