acd/infinitive #27: Direct control of FE4A via RS485 by writing to its registers

jskubick, 2020-01-27:
Is there experimental support anywhere in Infinitive for WRITING directly to the registers of something like a FE4A fan coil to do things like:

* Directly control the fan coil's blower speed. Given that the protocol apparently has its roots in some fairly old products, my guess is that the raw protocol probably isn't very abstracted, and the "fan control" register's bits probably have a direct, one-to-one correspondence with the state of specific pins on the ECM's wiring harness. It's also possible it might depend upon one or more analog voltage levels on the harness, probably determined by their own registers.

* Directly control the compressor's off/on, or off/low/high running state. My guess is that this can be done in two ways... by communicating directly with the condenser unit (if it supports Infinity/Evolution directly), or by asking the FE4A to do it as a proxy (if the condenser unit does NOT itself support Infinity/Evolution directly).

* Directly control the heat strips. I suspect this might be a two-bit value... one of which is analogous to closing or opening R to W/W2 on a normal system, and one of which is analogous to closing or opening R to W1 on a system that supports intelligent heating with an appropriate module (like the one that has 3k and 6k elements whose control works more or less the same way that a 3-way incandescent light bulb does).

* Controlling things like fan delay settings. My guess is that the FE4A probably uses the exact same ECM as the FV4C, and has the same 4 options and restrictions behind the scenes... probably set by either two bits wired to pins in the ECM header cable, or a register whose value specifies an analog voltage level for a pin on the cable.

Why: I'm about to buy a new central A/C. Up to now, I planned to get a FV4C, just because I already have so much knowledge about how its control scheme works and what it can do. However, I know that a FE4A is almost the same price, and can theoretically do more... but without the ability to directly control things like its blower speed, the FE4A's additional capabilities wouldn't do me much good. My entire goal is to implement my own dehumidification-optimizing thermostat controller building upon the work I did in the past for my old system (a Bryant 704b, which has more or less the same control scheme as a FV4).

Merely being able to set a cool-to-dehumidify value in a Cor using Infinitive wouldn't help, because Carrier's thermostats won't *allow* you to set target humidity levels below 45% (unless, of course, Infinitive bypasses Cor's UI checks, and allows you to set values outside the range the thermostat itself would allow you to set via its own UI?). I suppose I could change Cor's setpoint values on the fly to something like 66F and 90F to trick it into turning the system on and off, but that wouldn't necessarily solve the problem of needing to be able to change the fan speed... or to do things that Cor officially won't allow you to do, like cool-to-dehumidify with simultaneous electric reheat via the heat strips (which you'd trigger on a FV4C by closing R to Y and R to W to activate *both* the compressor *and* heat strips, while leaving R-G open to imply that the system should reduce the blower speed).

Assuming Infinitive's apparent lack of support for writing to registers isn't deliberate, and is just because nobody got around to doing it yet, I'd definitely be willing to help flesh out support for it if I could just find something resembling documentation for the FE4A's control registers (as noted above) to point me in the right direction. In any case, even knowing what to specifically search for on Google to find register documentation would probably be immensely helpful.

dulitz, 2021-08-07:
Did you ever make any progress on this?

jskubick, 2021-08-23:
No, I threw in the towel and bought a FV4C instead, then added some extra hardware of my own (temperature sensors on the coil and in the airflow below and above it, plus some relays to allow me to select lower/nominal/higher blower speed on the controller board (in addition to the Y/Y+Dh/Y+G control scheme)).

dulitz, 2021-08-23:
Cool. I'm finally beginning to do the documentation work that you asked for above. Hope you like your new system!