acd/infinitive #28: Code=exited

bryangauvin, 2020-02-20:
Im getting this error when starting the service. Followed your instructions step by step. Please help. Hate not having ecobee in my new house and this is the best thing I have found yet.

` infinitive.service - Infinitive Service
Loaded: loaded (/etc/systemd/system/infinitive.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Thu 2020-02-20 20:39:27 GMT; 66ms ago
Process: 8767 ExecStart=/root/go/bin/infinitive -httpport=8080 -serial=/dev/ttyUSB1 (code=exited, status=2)
Main PID: 8767 (code=exited, status=2)

Feb 20 20:39:27 raspberrypi systemd[1]: infinitive.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Feb 20 20:39:27 raspberrypi systemd[1]: infinitive.service: Failed with result 'exit-code'.

JohnZable, 2023-01-30:
Did you fix this?  I am having this issue also.

acd, 2023-11-16:
If you're using the serial device `/dev/tty0` while trying to talk to a USB serial adapter that's likely your problem. On my machine, the USB adapter shows up as `/dev/ttyUSB0`.