acd/infinitive #34: Muti-Zone and Schedule support

CptnWookie, 2022-02-02:
Just joined the Infinitive's team a year ago and since then I always wondered if there was a way to add multi-zone support as well as support for handling Schedules. 

I currently have a 2 Zones setup so if I can send you any data so this can be added, please let me know as this would be more versatile for those who don't have the Touch Thermostat.

acd, 2023-11-16:
Heya, if you're still interested I'm definitely curious to see what a two zone setup looks like. We may be able to grok what's different by observing data coming across the bus. If you can capture traffic for 10-15 seconds and share it that'd be a good start.

themrwookiee, 2023-12-31:
@acd I completely missed your response. I'm still interested for sure. How would you like me to capture the data ?

themrwookiee, 2024-01-26:
@acd If you'd like, I could test the release for you since I have 2 zones prior you make it available for everyone (if you need to)

madbrain76, 2025-01-29:
I have had a 10 zone system since 2010. It's actually 2 independent systems. One is 4 zones, another 6 zones.
I offered to help with the project back in 2017, at least to collect logs, but did not receive a response. I would very much like to get my systems working with Home assistant. It's one of the key pieces that's missing in my smart home at this time.
I know a little big of Golang, but am mostly a C/C++ developer.

themrwookiee, 2025-01-29:
> I have had a 10 zone system since 2010. It's actually 2 independent systems. One is 4 zones, another 6 zones.
> I offered to help with the project back in 2017, at least to collect logs, but did not [...]

Here's the fork I'm currently using as it supports multi-zones: https://github.com/lurgh/infinitive

gogades, 2025-02-07:
> I have had a 10 zone system since 2010. It's actually 2 independent systems. One is 4 zones, another 6 zones. I offered to help with the project back in 2017, at least to collect logs, but did not [...]

I only took over the maintenance of this project recently, but sadly my furnace died last month and I no longer have an Infinity/Connex compatible furnace.

That being said, I would be happy to help out in any way I can, or hand over the project to someone else.

madbrain76, 2025-02-15:
> Here's the fork I'm currently using as it supports multi-zones: https://github.com/lurgh/infinitive

Thank you. I will take a look !

madbrain76, 2025-02-16:
> I only took over the maintenance of this project recently, but sadly my furnace died last month and I no longer have an Infinity/Connex compatible furnace.
> 
> That being said, I would be happy to help out in any way I can, or hand over the project to someone else.

Very sorry to hear that. I would greatly appreciate some assistance. One is with the choice of USB to RS485 interface.
I'm looking at buying a pair of these for each of my Carrier systems :
https://www.amazon.com/gp/product/B081MB6PN2?psc=1

My question is about resilience. If the Pi3/Infinitive has an issue, such as software corruption (OS not booting) or worse, hardware (not powering up), what will happen to the Infinity systems ? If I understand correctly, RS485 is a serial bus. If power is cut to the USB interface, will all signals stop traveling, thus making the Infinity system inoperative ? And if so, would adding an external power source for it (say, USB hub) resolve the problem if the Pi3 is powered down, or software is not running?

madbrain76, 2025-03-10:
> I only took over the maintenance of this project recently, but sadly my furnace died last month and I no longer have an Infinity/Connex compatible furnace.
> 
> That being said, I would be happy to help out in any way I can, or hand over the project to someone else.

Thanks ! I had the installation done last Saturday on both furnaces, using a pair of Raspberry Pi 3B+. I have been spending a lot of time making an SD card that works. I just the infinitive daemon with my upstairs 6-zone system, and the web GUI seems to work great ! The only thing I noticed was wrong is that the humidity always shows 45% in every zone. It is also quite slow. Is that due to the slow communication to the Carrier bus, or the Pi 3B+ being insufficient ? It does not appear to hit >25% CPU usage in "vmstat 5".

I have not gotten to the MQTT and Home Assistant parts yet. It seems a bit daunting.

madbrain76, 2025-03-12:
> Unfortunately I am not an expert in programming or sysadmin, but I can tell you I have been powering my single zone system with a Raspberry Pi Zero W for many years with minimal resource utilization and uptime in tens of months. I use [Alpine Linux](https://alpinelinux.org) as the base because it runs fully from ram. You can even remove the SD card while operating with no issue.
> 
> I have my system wired to the control board in the air handler along side the Carrier SYSTXCCUIZ01 thermostat and use both in tandem with no issues. When the Pi is down, I lose online control, but thermostat maintains the 'communicating' operations of the system.

Thanks. It looks like the performance issues were showing up when I first set it up, but perf is OK now.

What's not OK is the fact that I can only effectively use one of my 2 systems at a time.
I am using the fork at https://github.com/lurgh/infinitive/, and posted the following issue 
https://github.com/lurgh/infinitive/issues/5

I believe your fork at https://github.com/gogades/hass-infinitive/tree/master does not support multi-zones, so I cannot use it, unfortunately.