acd/infinitive #37: Current Temperature Decimal Point

FSFirenze, 2024-05-09:
Good morning, this is not an issue report, instead a request for information.
I am wrestling the idea of adding RS485 monitoring to my currrent set-up (raspberryPIv->vHomeAssistant->vHomAssistant-infinitude -> Infinitude -> Carrier Infinity Thermostat.

One key feature that I am looking at is the ability to retrieve the Thermostat Current Temperature with at least one decimal point of resolution. With my current set-up i can only "see" zero decimal points.

Is the infinitude RS485 monitor capable of accessing the Thermostat Current Temperature with a resolution of 1 decimal point (r.g. 73.3degF ot 21.2degC)?

Thnkas

nebulous, 2024-05-13:
I'm interjecting slightly outside of my wheelhouse here, but from what I've observed the thermostat doesn't provide any higher resolution data than deg F. Some devices such as heat pumps do send data with higher granularity (expressed in 16ths of deg F if I recall) but the thermostat itself doesn't publish temperature(indeed doesn't need to since it's the controller).  Perhaps someone with a(4 wire) secondary temperature unit could show what that pushes to the bus. Ideally it would be less coarse (and would allow all of us to simulate other temperature sensors using inexpensive microcontrollers).

FSFirenze, 2024-05-14:
Thank you for your interjection!
So, I think we are saying that we definitely don't see the higher resolution on the RS485 bus, at least not in the basic 1-zone setup/
I think you are opening up to the fact that auxiliary temperature sensors may publish higher resolution temp, but it seems that the hope is related to the temperature measured by these aux sensors and not by the current temp measured by the main thermostat, please let me know if i am misunderstanding.

I may be willing to test out installing an auxiliary thermometer to help checking what we can see on the bus .... but I'll need some guidance on the model that I should invest on. I have a SYSTXCCITC01-B